rogerclarkmelbourne/STM32duino-bootloader #134: Unknown USB Device (Device Descriptor Request Failed) on STM32F103C8 Custom Board with STM32duino DFU Bootloader

ae-circuit, 2024-11-07:
Description: I'm facing an issue with my STM32F103C8 custom board. After flashing the STM32duino DFU bootloader, the device is recognized as an "Unknown USB Device (Device Descriptor Request Failed)" on Windows.

**Board Specifications:**

MCU: STM32F103C8
External Oscillator: 16MHz
Custom Board Design: Yes

**_Steps Taken:_**
**Bootloader Compilation and Flashing:**
I downloaded the STM32duino DFU bootloader from this repository.
Modified the code to set the external oscillator to 16MHz in the hardware.h file.
Compiled the bootloader using make generic-pc13.
Successfully flashed the bootloader onto the board using an ST-Link V2 programmer.

**After Flashing:**
When I connect the board via USB, Windows does not recognize it correctly. It shows "Unknown USB Device (Device Descriptor Request Failed)" in Device Manager.

**Troubleshooting Attempts:**
Checked the USB wiring on my custom board.
Reflashed the bootloader to ensure no errors occurred during flashing.
Verified that the 16MHz oscillator is properly connected and providing a stable clock signal.
Attempted to compile with different configurations in case it was a build issue.
Relevant Code Modifications: In hardware.h, I ensured the following settings to match the 16MHz external oscillator:

**System Information:**
Host OS: Windows 10
USB Drivers: STM32 USB drivers for DFU gets detected when STM32 BluePill is flashed with relevant DFU bootloader.
Request for Assistance: Could you please advise if there are additional settings or configurations required to ensure USB recognition, or if there might be another cause for this issue? Any guidance on additional steps or troubleshooting would be greatly appreciated.

Thank you for your assistance.

rogerclarkmelbourne, 2024-11-08:
This isn't an issue with the unmodified code.

There are many possible reasons why your board is not working, including incorrect clock source setting, incorrect PLL chain values, or possibly even hardware problems as you have a custom PCB

Build a project using the STM32Cube IDE that has USB and get that working first, and then use those PLL settings etc in this bootloader

ae-circuit, 2024-11-08:
I did write a program using STM32CubeIDE with virtual com port and it does get detected by the system when connected. But when the STM32Duino bootloader is flashed, it shows an unknown USB after being detected by the Windows. Isn't it possible to have a different repo or different folder for .bin files created after setting all things for XTAL16M? I created it but need to confirm if my steps are correct or not. 

Anyways that will be helpful if you share what exact things I need to troubleshoot.

rogerclarkmelbourne, 2024-11-08:
Just to confirm you changed the #define to specify 16MHz crystal ?