rogerclarkmelbourne/STM32duino-bootloader #135: upload arduino code in stm32f103c8 after

shehabhassan, 2024-11-17:
here is my device manager recognize the STM32F103C8 as Maple Serial (COM13) 
[image]
und here my arduino IDE screenshot when i tried to upload the code it doesn't find the Device every Uploading .

[image]

how i can solve this issue ? 
i have struggle with this issue for three days.  

Notes: I followed the instruction of flashing bootloader as provided in this link here : 
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Flashing-Bootloader-for-BluePill-Boards

thanks in advance

fpistm, 2024-11-17:
You have to enable USB CDC and use the com port created.

shehabhassan, 2024-11-18:
> You have to enable USB CDC and use the com port created.

Okay Like this one Here : 

[image]

fpistm, 2024-11-18:
you have to flash it first, then select the new com port created.

shehabhassan, 2024-11-18:
> you have to flash it first, then select the new com port created.

do you mean flash arduino bootloader ?

shehabhassan, 2024-11-18:
> you have to flash it first, then select the new com port created.

could you please clarify your suggestion because i am still confused and struggled at the position ?

fpistm, 2024-11-18:
As stated here: https://github.com/stm32duino/Arduino_Core_STM32/wiki/Upload-methods#maple-dfu-bootloader

> [!WARNING]
> USB CDC have to be enable else you will not be able to upload automatically as bootloader reset sequence are managed through the USB CDC communication port.

Your issue is that the board is not started in Bootloder mode and the COM port is the USART one not the CDC. So the reset sequence to start un BL mode can't be done.

To start in BL mode manualy set the BOOT0 at 1 or eraser fully the flash then flash the bootloader, as no application it will stay in BL mode (having a DFU device and no com port).
Then flash the apllication with CDC enabled.
Next flash will work using the CDC com port.

shehabhassan, 2024-11-19:
[image]
here is my device manager 

and the monitor display this message : 
[image]

after that i followed you instruction to enable CDC and Select the right port number. 
[image]

[image]

> As stated here: https://github.com/stm32duino/Arduino_Core_STM32/wiki/Upload-methods#maple-dfu-bootloader
>
> Warning
>
> USB CDC have to be enable else you will not be able to upload automatically as [...]

fpistm, 2024-11-19:
Com port is only used to reset in Maple DFU mode.
As said after the first flashing of the bootloader, your board should be in this mode.
[image]
But it requires a dedicated driver available here:
https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/drivers/win

More info here: https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/stm32duino-bootloader
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Flashing-Bootloader-for-BluePill-Boards